copterust/vl53l0x #13: Compilation error `error[E0277]`

mobergmann, 2024-07-20:
I tried to use this library for using my vl53loc-v2 senor. But when I try to compile the project I get an compilation error for this library:
```
$ cargo build
   Compiling vl53l0x v1.0.0
error[E0277]: the trait bound `u16: cast::From<_>` is not satisfied
   --> /home/mobergmann/.cargo/registry/src/index.crates.io-6f17d22bba15001f/vl53l0x-1.0.0/src/lib.rs:155:17
    |
155 |         Ok((u16(buffer[0]) << 8) + u16(buffer[1]))
    |             --- ^^^^^^^^^ the trait `cast::From<_>` is not implemented for `u16`
    |             |
    |             required by a bound introduced by this call
    |
    = help: the trait `cast::From<f64>` is implemented for `f32`
    = help: for that trait implementation, expected `f32`, found `u16`
note: required by a bound in `u16`
   --> /home/mobergmann/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cast-0.2.7/src/lib.rs:181:1
    |
181 | fns!(f32, f64, i8, i16, i32, i64, isize, u8, u16, u32, u64, usize);
    | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^---^^^^^^^^^^^^^^^^^^
    | |                                            |
    | |                                            required by a bound in this function
    | required by this bound in `u16`
    = note: this error originates in the macro `fns` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0277]: the trait bound `u16: cast::From<_>` is not satisfied
   --> /home/mobergmann/.cargo/registry/src/index.crates.io-6f17d22bba15001f/vl53l0x-1.0.0/src/lib.rs:155:13
    |
155 |         Ok((u16(buffer[0]) << 8) + u16(buffer[1]))
    |             ^^^^^^^^^^^^^^ the trait `cast::From<_>` is not implemented for `u16`
    |
    = help: the trait `cast::From<f64>` is implemented for `f32`
    = help: for that trait implementation, expected `f32`, found `u16`

error[E0277]: the trait bound `u16: cast::From<_>` is not satisfied
   --> /home/mobergmann/.cargo/registry/src/index.crates.io-6f17d22bba15001f/vl53l0x-1.0.0/src/lib.rs:155:40
    |
155 |         Ok((u16(buffer[0]) << 8) + u16(buffer[1]))
    |                                    --- ^^^^^^^^^ the trait `cast::From<_>` is not implemented for `u16`
    |                                    |
    |                                    required by a bound introduced by this call
    |
    = help: the trait `cast::From<f64>` is implemented for `f32`
    = help: for that trait implementation, expected `f32`, found `u16`
note: required by a bound in `u16`
   --> /home/mobergmann/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cast-0.2.7/src/lib.rs:181:1
    |
181 | fns!(f32, f64, i8, i16, i32, i64, isize, u8, u16, u32, u64, usize);
    | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^---^^^^^^^^^^^^^^^^^^
    | |                                            |
    | |                                            required by a bound in this function
    | required by this bound in `u16`
    = note: this error originates in the macro `fns` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0277]: the trait bound `u16: cast::From<_>` is not satisfied
   --> /home/mobergmann/.cargo/registry/src/index.crates.io-6f17d22bba15001f/vl53l0x-1.0.0/src/lib.rs:155:36
    |
155 |         Ok((u16(buffer[0]) << 8) + u16(buffer[1]))
    |                                    ^^^^^^^^^^^^^^ the trait `cast::From<_>` is not implemented for `u16`
    |
    = help: the trait `cast::From<f64>` is implemented for `f32`
    = help: for that trait implementation, expected `f32`, found `u16`

For more information about this error, try `rustc --explain E0277`.
error: could not compile `vl53l0x` (lib) due to 4 previous errors
```

My project is basically straight out of the cargo generate tool for the Arduino Uno. I customized it a bit, but basically it looks like this:
```rust
    let mut tof = vl53l0x::VL53L0x::new(i2c).expect("tof");
    tof.set_measurement_timing_budget(200000).expect("time budget");
    tof.start_continuous(0).expect("start");

    // let duration = Milliseconds::<u64>(embedded_time::duration::ms);

    loop {
        led.toggle();
        
        let mls = tof.read_range_continuous_millimeters();
        ufmt::uwriteln!(&mut serial, "Distance: {:?}?", mls).unwrap();
    }
```

mobergmann, 2024-07-24:
I just noticed, that the error is not occurring with the stable rust version:
```
[nix-shell:~/Source/vl53l0x]$ cargo --version
cargo 1.77.1

[nix-shell:~/Source/vl53l0x]$ rustc --version
rustc 1.77.2 (25ef9e3d8 2024-04-09) (built from a source tarball)
```

but only with the nightly version. Sadly, afaict is the nightly version mandatory for my project.

little-arhat, 2024-07-24:
@mobergmann thanks for reporting, we will look into supporting nightly (and will be happy to review proposed changes)

mobergmann, 2024-07-24:
I am currently looking into it and it seems, that the issue is in the [cast](https://docs.rs/cast/latest/cast/) crate.

mobergmann, 2024-07-24:
I created an issue on the cast repository: https://github.com/japaric/cast.rs/issues/46

little-arhat, 2024-07-24:
ah, we should just get rid of `cast` imo -- does not bring a lot of value.
We did this in few other crates, eg https://github.com/copterust/mpu9250/commit/55f7e6e6d25441b6f007cad64c1fa0ea8a96fef3

mobergmann, 2024-07-24:
I could try doing this, but I am fairly new to "real world" rust. Therefore my code might be a bit ... clumsy. I will focus strongly on the code you referenced.

mobergmann, 2024-07-24:
@little-arhat I created an MR #14 with the cast crate removed. I hope the code is okay.

PR #14 (2024-07-24): Removed the cast crate
fixes #13
closed 2024-07-24 by little-arhat