copterust/vl53l0x #15: How to share the i2c bus?

mobergmann, 2024-07-25:
Currently the I2C bus is consumed by the vl53l0x driver. However, I want to use multiple drivers on the i2c bus, which is not possible after the I2C has been consumed. It seems, that the I2C cannot be copied. I am therefore a bit riddled on how to solve this issue. Is there sth I am missing? I read a bit and there seems to be the crate [shared-bus](https://docs.rs/crate/shared-bus/latest), which tries to solve this issue by providing a bus handler, but it seems to be not compatible with this library(?). How do you solve this issue?

mobergmann, 2024-07-25:
Okay, I found out, that I can use the [embedded_hal_bus](https://docs.rs/embedded-hal-bus/0.2.0/embedded_hal_bus/index.html) crate, especially the [i2c](https://docs.rs/embedded-hal-bus/0.2.0/embedded_hal_bus/i2c/index.html) module, to share the bus with many devices.

closed 2024-07-25 by mobergmann

little-arhat, 2024-07-25:
indeed, we have example with shared bus [here](https://github.com/copterust/proving-ground/tree/master/shared_i2c), but it's quite old, so maybe there are better ways now.